uzh-rpg/rpg_dvs_ros #10: Compiling error of the dvs_ros_driver

doomzzju, 2016-07-31:
Hi guys, I'm trying to compile the rpg_dvs_ros package these day. When I compiled the dvs_ros_driver, there is a error just like below:

/opt/ros/indigo/lib/libcamera_calibration_parsers.so: undefined reference to `YAML::detail::node_data::convert_to_map(boost::shared_ptr<YAML::detail::memory_holder>)'

How can I fix this bug? thanks~~

eliasm, 2016-08-09:
Could you give some more details? Which package causes the error? What version of ROS and Ubuntu are you using?

doomzzju, 2016-08-09:
I use ROS indigo and Ubuntu 14.04. The  package dvs_ros_driver causes this error, it seems like something about YAML?

eliasm, 2016-12-01:
Are you also using the `yaml_cpp_catkin` package from ETH/ASL? If so, please update that package to the latest version, clean your entire workspace, and build everything again.

closed 2017-02-01 by eliasm