uzh-rpg/rpg_dvs_ros #12: Calibration Pattern

Monopolymer, 2016-11-30:
Hello,
i have a DVS128 from inilabs.
Is it possible to acquire  a calibration pattern somewhere or do i have to build one myself ?

eliasm, 2016-12-01:
As far as I know, you cannot buy such a pattern anywhere, but it should not be too difficult to build that yourself with, e.g., an [arduino](https://www.arduino.cc/en/tutorial/blink) Otherwise, you can use your screen to show a blinking pattern, as described here in Section III: http://rpg.ifi.uzh.ch/docs/IROS14_Mueggler.pdf

closed 2016-12-01 by eliasm

Monopolymer, 2016-12-01:
Thank you for your reply. I will try it. I did read somewhere else that someone managed to calibrate a Davis Camera with a checkboard would hat work as well?

eliasm, 2016-12-01:
Yes, the DAVIS also outputs conventional image frames that you can use for standard camera calibration. Since the same physical pixel is used for both the frames and the events, the image-based calibration also holds for the events. We only use the LEDs for the DVS128; for the DAVIS we use the images.

Monopolymer, 2016-12-02:
Thank you again. I have a few mor questions.
Can I use white led for my calibration pattern or schould i use infrared led?
I guess the led schould all blink at the same time.
Is the background colour of the led mount of any importance?
I will try and build one this weekend.
I apreciate your help verry much.

eliasm, 2016-12-05:
Yes, they should blink at the same time. I would go for visible LEDs, but infrared might work too. The background should not be reflective, but otherwise it does not matter.

Monopolymer, 2016-12-14:
I made a5x5 pattern with an arduino. It worked like a charm thank you again.