uzh-rpg/rpg_dvs_ros #22: branch master: [davis_ros_driver-1] process has died

gussmith, 2017-03-29:
Recent changes after install on new computer causes the Davis driver to die right upon starting dvs_renderer:
roslaunch dvs_renderer davis_mono_events_only.launch
or 
roslaunch dvs_renderer davis_mono.launch
[davis_mono_nodelet.launch does work though]
[davis_ros_driver-1] process has died [pid 27252, exit code -7, cmd /home/gus/catkin_rpg_master/devel/lib/davis_ros_driver/rdavis_ros_driver __name:=davis_ros_driver __log:=/home/gus/.ros/log/76c37878-128e-11e7-a827-d0577b49371b/davis_ros_driver-1.log].
log file: /home/gus/.ros/log/76c37878-128e-11e7-a827-d0577b49371b/davis_ros_driver-1*.log
The log file remains empty.

luccide, 2017-04-18:
Having the exact same issue here

eliasm, 2017-04-18:
Thanks for reporting this issue. Could you please check whether #24 resolves the issue?

llongi, 2020-09-11:
Fix merged years ago.

closed 2020-09-11 by llongi