uzh-rpg/rpg_dvs_ros #23: Compilation error on catkin build

lokijota, 2017-04-04:
I'm getting this on Ubuntu 14.04, with gcc 4.9:

Starting  >>> davis_ros_driver
_______________________________________________________________________________
Errors     << davis_ros_driver:make /home/mg/logs/davis_ros_driver/build.make.000.log
make[2]: *** No rule to make target `/home/mg/devel/.private/davis_ros_driver/lib/libcaer.so', needed by `/home/mg/devel/.private/davis_ros_driver/lib/davis_ros_driver/davis_ros_driver'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/davis_ros_driver.dir/all] Error 2
make: *** [all] Error 2
cd /home/mg/build/davis_ros_driver; catkin build --get-env davis_ros_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << davis_ros_driver:make                [ Exited with code 2 ]

ANy ideas? newbie into all of this, a bit lost.

luccide, 2017-04-18:
I have the same problem, which i can solve by pointing the cmake file at the libcaer.so directly. but after doing this i seem to have the davis driver crash whenever i try to run it.

eliasm, 2017-04-18:
We are currently looking into this issue. Can you try to run `catkin config --merge-devel` before `catkin build` and see whether that resolves the issue?

luccide, 2017-04-19:
Thanks elisam, 
using catkin config --merge-devel prevented catkin build from working with the error

 "Error: The current devel space layout, `linked`,is incompatible with the configured layout, `merged`."

However when built with catkin_make everything works perfectly now. 

Thanks

closed 2017-04-19 by eliasm