uzh-rpg/rpg_dvs_ros #33: failing to build dvs_ros_driver

fabianmue, 2017-05-15:
After following the steps in the readme, invoking catkin build dvs_ros_driver always fails for me.

Here is the terminal output (including my catkin workspace config):

~/catkin_ws/src$ catkin build dvs_ros_driver
-----------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/indigo
Workspace:                   /home/fabian/catkin_ws
-----------------------------------------------------------
Source Space:       [exists] /home/fabian/catkin_ws/src
Log Space:         [missing] /home/fabian/catkin_ws/logs
Build Space:        [exists] /home/fabian/catkin_ws/build
Devel Space:        [exists] /home/fabian/catkin_ws/devel
Install Space:      [unused] /home/fabian/catkin_ws/install
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '9' packages in 0.0 seconds.                                                                                                                                        
[build] Updating package table.                                                                                                                                                   
Starting  >>> catkin_simple                                                                                                                                                       
Finished  <<< catkin_simple                 [ 1.1 seconds ]                                                                                                                       
Starting  >>> dvs_msgs                                                                                                                                                            
Starting  >>> libcaer_catkin                                                                                                                                                      
Finished  <<< dvs_msgs                      [ 2.0 seconds ]                                                                                                                       
__________________________________________________________________________________________________________________________________________________________________________________
Errors     << libcaer_catkin:make /home/fabian/catkin_ws/logs/libcaer_catkin/build.make.000.log                                                                                   
Cloning into 'libcaer_src'...
remote: Counting objects: 5238, done.
remote: Compressing objects: 100% (58/58), done.
remote: Total 5238 (delta 34), reused 0 (delta 0), pack-reused 5180
Receiving objects: 100% (5238/5238), 11.10 MiB | 2.28 MiB/s, done.
Resolving deltas: 100% (4192/4192), done.
Checking connectivity... done.
Note: checking out '57466910da84c25231aab57e99f246239722a628'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at 5746691... Events: add Point4D event, useful for tracking homogeneous coordinates for example.
make[3]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
/home/fabian/catkin_ws/build/libcaer_catkin/libcaer_src-prefix/src/libcaer_src/src/log.c: In function ‘caerLog’:
/home/fabian/catkin_ws/build/libcaer_catkin/libcaer_src-prefix/src/libcaer_src/src/log.c:29:6: warning: stack protector not protecting local variables: variable length buffer [-Wstack-protector]
 void caerLog(uint8_t logLevel, const char *subSystem, const char *format, ...) {
      ^
/home/fabian/catkin_ws/build/libcaer_catkin/libcaer_src-prefix/src/libcaer_src/src/frame_utils.c: In function ‘caerFrameUtilsWhiteBalance’:
/home/fabian/catkin_ws/build/libcaer_catkin/libcaer_src-prefix/src/libcaer_src/src/frame_utils.c:648:54: warning: unused parameter ‘framePacket’ [-Wunused-parameter]
 void caerFrameUtilsWhiteBalance(caerFrameEventPacket framePacket) {
                                                      ^
In file included from /home/fabian/catkin_ws/build/libcaer_catkin/libcaer_src-prefix/src/libcaer_src/include/frame_utils.h:16:0,
                 from /home/fabian/catkin_ws/build/libcaer_catkin/libcaer_src-prefix/src/libcaer_src/src/frame_utils.c:1:
/home/fabian/catkin_ws/build/libcaer_catkin/libcaer_src-prefix/src/libcaer_src/src/frame_utils.c: In function ‘caerFrameUtilsDemosaic’:
/home/fabian/catkin_ws/build/libcaer_catkin/libcaer_src-prefix/src/libcaer_src/include/events/frame.h:164:5: warning: assuming signed overflow does not occur when simplifying conditional to constant [-Wstrict-overflow]
  if (n < 0 || n >= caerEventPacketHeaderGetEventCapacity(&packet->packetHeader)) {
     ^
make[3]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
make[2]: *** No rule to make target `/home/fabian/catkin_ws/devel/lib/libcaer.so', needed by `/home/fabian/catkin_ws/devel/lib/liblibcaer_catkin.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/libcaer_catkin.dir/all] Error 2
make: *** [all] Error 2
cd /home/fabian/catkin_ws/build/libcaer_catkin; catkin build --get-env libcaer_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................................................................................................................
Failed     << libcaer_catkin:make           [ Exited with code 2 ]                                                                                                                
Failed    <<< libcaer_catkin                [ 13.3 seconds ]                                                                                                                      
Abandoned <<< dvs_ros_driver                [ Unrelated job failed ]                                                                                                              
[build] Summary: 2 of 4 packages succeeded.                                                                                                                                       
[build]   Ignored:   5 packages were skipped or are blacklisted.                                                                                                                  
[build]   Warnings:  None.                                                                                                                                                        
[build]   Abandoned: 1 packages were abandoned.                                                                                                                                   
[build]   Failed:    1 packages failed.                                                                                                                                           
[build] Runtime: 14.5 seconds total.  

I am using Ubuntu 14.04 and I have updated gcc and g++ to 4.9. I have never used the package before, am I missing something obvious?

togaen, 2017-06-22:
Could you try making the changes in this pull request and retry building?

https://github.com/uzh-rpg/rpg_dvs_ros/pull/38

fabianmue, 2017-06-23:
successfully built the new master with the changes from #38

closed 2017-06-23 by fabianmue