uzh-rpg/rpg_dvs_ros #36: Error when building davis_ros_driver

pertierre, 2017-06-21:
I'm using Ubuntu 14.04 with gcc 4.9 and I'm getting exactly the same issue as here: https://github.com/uzh-rpg/rpg_dvs_ros/issues/23

However, even if I ran `catkin config --merge-devel` before `catkin build davis_ros_driver` as it was advised I still having the same problem.
Has anyone any idea of what I can try?

togaen, 2017-06-22:
Could you try making the changes in this pull request https://github.com/uzh-rpg/rpg_dvs_ros/pull/38 and retry building?

pertierre, 2017-06-23:
Actually this link target was already removed in davis_ros_driver/CMakeLists.txt, but I tried to remove it in dvs_ros_driver and build. However I'm still getting the same error (not rule to make '/home/.../lib/libcaer.so' needed by '/home/.../lib/liblibcaer_catkin.so').

eliasm, 2017-06-23:
Which version of ROS are you using? Did you try to clean your workspace before building with `catkin clean`?

pertierre, 2017-06-23:
Yes, I tried `catkin clean` also. I'm using indigo.

togaen, 2017-06-23:
Is this your actual error message?

`no rule to make '/home/.../lib/libcaer.so' needed by '/home/.../lib/liblibcaer_catkin.so`

That's a different problem than #23. Something else must be happening. The path `/home/../lib/liblibcaer_catkin.so` does not seem right... it probably shouldn't be `liblibcaer_catkin.so`. Are there any modifications to any cmake files? Have you sourced your workspace after cleaning?

charbelazzi8, 2017-06-26:
I am actually having the same exact issue. I agree with togaen about the path is weird. I used the new commit from #38 and source the workspace correctly but still the same issue with both catkin_make and catkin build. I believe something is missing in the CMakeLists.txt. Any updates regarding this problem? Thanks

togaen, 2017-06-26:
@charbelazzi8 are you also on Indigo? Or what version of ROS? I don't think patch #38 is any help for the weird path issue, unfortunately.

charbelazzi8, 2017-06-26:
Yes I am using Indigo.

togaen, 2017-06-26:
Interesting. @pertierre was also using Indigo; maybe there is something about catkin in Indigo that is not playing nice with this package. Are you unable to upgrade to Kinetic?

charbelazzi8, 2017-06-27:
I am not currently able to upgrade to Kinetic but I will try to find away around this problem. It is mostly a linking problem so it should be fixable. I will update you when I find a solution

pertierre, 2017-06-30:
@togaen yes, you are right, it has this difference compared with problem #23.
I have a friend who is using Kinetic and had no problem to install the drivers, so probably this issue only happens on Indigo. Unfortunately I am unable right now to upgrade either, as I am working with Ubuntu 14.04.

cJarvers, 2017-07-21:
Hello,
I am also using Ubuntu 14.04 and experienced the same error as @pertierre (installation on a different computer with Ubuntu 17.04 worked fine, with 14.04 I got the error `'/home/.../lib/libcaer.so' needed by '/home/.../lib/liblibcaer_catkin.so'`). Since I can currently not update to 16.04, I delved into the issue a little further and found a workaround. Basically, I discovered the following things:

- the doubling of "lib" in the library name is due to the `cs_add_library()` command in line 21 of [libcear's CMakeLists.txt](https://github.com/uzh-rpg/rpg_dvs_ros/blob/master/libcaer_catkin/CMakeLists.txt). By replacing `${PROJECT_NAME}` with `caer_catkin` and doing the same in the subsequent two lines, I solved the naming issue. However,
- the doubling of "lib" was not the main problem. Apparently, the libcaer shared object filed got installed to `${CATKIN_DEVEL_PREFIX}/lib/x86_64-linux-gnu/`, whereas line 24 of the CMakeLists looks for them in `${CATKIN_DEVEL_PREFIX}/lib/`. I extended the path on that line and was able to install libcaer_catkin that way.
- the other packages (dvs_ros_driver etc.) still look for libcaer under the wrong path. Therefore, after installing libcaer_catkin, I added a symbolic link to `${CATKIN_DEVEL_PREFIX}/lib/x86_64-linux-gnu/libcaer.so` at  `${CATKIN_DEVEL_PREFIX}/lib/libcaer.so`. Afterwards, I was able to install all other packages as well.

I tested the installation by connecting a DVS and running the DVS renderer. The installation seems to work. But probably there is a more principled way to do this than with this workaround.

pertierre, 2017-07-21:
Really thank you a lot, @cJarvers! I followed your steps and I don't know if I understand what you did in your last step, but it worked fine to me when I changed the path to `@CATKIN_DEVEL_PREFIX@/lib/x86_64-linux-gnu/libcaer${CMAKE_SHARED_LIBRARY_SUFFIX})` in the file `libcaer_catkin/cmake/libcaer-extras.cmake.in`.

iliis, 2017-10-02:
Thanks @cJarvers! I had the samme issue using Kinetic on Fedora 26 (I built ROS from source).
Changing `${PROJECT_NAME}` to `caer_catkin` and adding a symlink from `/home/.../lib/libcaer.so` to /home/.../lib64/libcaer.so` was enough to compile it.
To run the renderer another symlink was required for `libcaer.so.2`

llongi, 2020-09-11:
Official libcaer build is now used, see README for further details.

closed 2020-09-11 by llongi