uzh-rpg/rpg_dvs_ros #41: Issue with loading events from ROS bag file into MATLAB ROS Toolbox

sneheshs, 2017-07-11:
Hi,

We have ROS bag files with IMU, image, and events data from DAVIS 240B. Everything else loads file with rosbag function. However, when trying to decode /dvs/events messages the toolbox is not able to handle it as it seems ros bag stores it in custom data type. We used the dvs_msg folder with rosgenmsg function in the toolbox to generate custom data support. However, this step does not seem to work due to the EventArrayRule.bmr file defined in the package.xml. MATLAB ROS toolbox is not able to handle this. Can you please advise how we might be able to get this hurdle?

Best,
Snehesh

eliasm, 2017-07-12:
The `EventArrayRule.bmr` is only needed to convert old (2 years+) bag files, so you might try to remove it from the `package.xml` and try again.

sneheshs, 2017-07-25:
Thanks for the clarification. It turns out the instructions in the Matlab rosgenmsg has a little trick to it that is not documented. Not sure it is a bug or not but got it to work after some trial and error. The trick turns out that when you call rosgenmsg('dvs_msgs') the folder structure is expected in the following way:

```
<Your_Working_Folder>
|--- dvs_msgs
|...  |--- dvs_msgs
|...  |...  |--- msg
|...  |...  |...  |--- Event.msg
|...  |...  |...  |--- EventArray.msg
|...  |...  |--- package.xml

```
Please note the dvs_msgs folder needs to be inside another dvs_msgs folder. Hopefully, this will help others from a little bit of a headache.

Best,
Snehesh