uzh-rpg/rpg_dvs_ros #50: DAVIS build broken

sneheshs, 2017-10-19:
Hi,

It looks like after adding DAVIS_CONFIG_DVS_FILTER_..... the build is broken. I pulled the latest of the libcaer, however, that did not help. I went back to a few commits and it works fine. I did not validate exactly which step it broke but I checkout 4c1e1ee00c4b87fa0280344aa1b87d572ad9ee9a commit and it seems to be working fine. From the code, it looks like the changes were due to support for miniDVS so I went to that commit. Please fix or let us know if there is an extra step I might be missed to get over this issue with the latest commit.

Best,
Snehesh

supitalp, 2017-10-20:
Hi Snehesh,
Can you checkout the master branch of the davis_ros_driver (and make sure it is up to date), and try the following?

```
cd path_to_your_catkin_workspace
rm -rf build/libcaer_catkin/ build/davis_ros_driver/ devel/lib/libcaer.so devel/lib/davis_ros_driver/
catkin build --force-cmake libcaer_catkin davis_ros_driver
```

Please let us know if that works, or what error message you get if that does not.

Best,
Henri

nicolaroberto-tesla, 2017-10-31:
Hi there, 

i've checked it out myself.

Still does not build with the indicated workaround. The output error is:

```
make[2]: *** No rule to make target '/home/vagrant/catkin_ws/devel/lib/libcaer.so', needed by 
'/home/vagrant/catkin_ws/devel/lib/liblibcaer_catkin.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/libcaer_catkin.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

Cordialement,

Nicola

supitalp, 2017-10-31:
Hi,
The error looks like this one: https://github.com/uzh-rpg/rpg_dvs_ros/issues/36
Have you tried using the fix given by cJarvers there? Can you tell us a bit more about your system (which linux distribution / version, which version of ROS)?

Also, if that does not work, have you tried checking out commit https://github.com/uzh-rpg/rpg_dvs_ros/commit/4c1e1ee00c4b87fa0280344aa1b87d572ad9ee9a (as sneheshs), clean your build and devel folders as indicated above, and rebuild it? Does that solve the issue?

Thanks,
Henri

sneheshs, 2017-10-31:
Hi Henri and Nicola,

I had to revert back to 4c1e1ee commit for this to work. However, please make sure you update to the latest caer library as well. I needed to do that for this to work.

Best,
Snehesh

nicolaroberto-tesla, 2017-10-31:
Hi There,

I am on an up-to-date ArchLinux, ros LUNAR installed through package manager and works with other stuff.

I have managed to get past the libcaer compilation by applying the fix of cJarvers you mentioned (modifying libcaer's `CMakeLists.txt` and `libcaer-extras.cmake.in`, no symlinks).

The fix both works with `HEAD` and `4c1e1ee` (the libcaer part).

However, afterwards, `catkin build` gets stuck with `davis-ros-driver` as it does not find the `rospkg` python module. 

I know there are issues with ArchLinux being built around python3 and conflicting with ROS but the workaround suggested in ArchWiki (forcing python2, that works with all the other packages i tried) seems to be ignored by rpg_dvs_ros.

I'm looking into it.

Cordialement,

Nicola R.

nicolaroberto-tesla, 2017-11-04:
Hi There,

I've fixed the python problem for ArchLinux.

To get rid of missing package error, I just forced `dvs_ros_driver` to use python2 by modifying `rpg_dvs_ros/dvs_ros_driver/cfg/DVS_ROS_Driver.cfg`.

In the end it compiled. I did not test it because the DAVIS camera is at the lab.

When I tried to compile instead dvs_rendered, problems started to rain down.

To make it compile and, most important, run, I had to:

symlink all the references to:

- libopencv_core.so.3.3.0
- libopencv_imgproc.so.3.3.0
- libopencv_imgcodecs.so.3.3.0

to their 3.3.1 version.

Also, I had to symlink

 - libPocoFoundation.so.48

to its .49 version.

After unsuccesfully searching how to modify these "requests for libraries" on the `rpg_dvs_ros` side, I gave up and grabbed a sandwich.

On another system, building the dvs_rendered produced the same kind of problems of before (wrong library version), but with libboost.

After this I really gave up.

Best,

nicolaroberto-tesla, 2017-11-07:
Tried again, camera is "busy" so no driver tests.
No luck with the rendererer though.
Still gives back a "attempted to unlock mutex that was not locked" and dies (exit code -6).
At this point IDK if the problem is the code or in the wrong libraries.
Best,

sneheshs, 2017-11-07:
Just for the sake of validating your steps vs Arch Linux related issues, I'd recommend trying it out in an Ubuntu setup.

nicolaroberto-tesla, 2017-11-08:
In *buntu it works out of the box.

Best,

nicolaroberto-tesla, 2017-11-11:
Thanks supitalp,

I've tried it, no out-of-date stuff found. 

It was less time consuming to bring up a *buntu vm and now it happily works.

Best,

llongi, 2020-09-11:
Official libcaer build is now used, see README for further details.

closed 2020-09-11 by llongi