uzh-rpg/rpg_dvs_ros #57: Occasional DVS timestamps with negative delta time about 99ms

tobidelbruck, 2018-01-31:
In developing the jAER rosbag reader, I notice in one recording from an advanced Davis346 sensor that in the middle of ROS messages holding topic "/dvs/events" there are events with negative delta times around 99ms. They occur in the middle of messages. Samples below. Does anyone know if this might be a consequence of the fact that the rosbag might not be written in strict monotonic Davis timestamp order, or more likely from a flaky USB connection?

If I discard these events, then the next event is apparently always newer. I.e. it doesn't appear that there is a jump backwards in time of 99ms of the entire stream (however, to be confirmed).

`WARNING: Discarding event with nonmonotonic timestamp detected for event 16106 in this message; delta time=-998999
Jan 31, 2018 8:35:14 AM net.sf.jaer.eventio.ros.RosbagFileInputStream getNextRawPacket
WARNING: Discarding event with nonmonotonic timestamp detected for event 4625 in this message; delta time=-998997
Jan 31, 2018 8:35:14 AM net.sf.jaer.eventio.ros.RosbagFileInputStream getNextRawPacket
WARNING: Discarding event with nonmonotonic timestamp detected for event 2827 in this message; delta time=-998968
Jan 31, 2018 8:35:15 AM net.sf.jaer.eventio.ros.RosbagFileInputStream getNextRawPacket
WARNING: Discarding event with nonmonotonic timestamp detected for event 4860 in this message; delta time=-998991
Jan 31, 2018 8:35:16 AM net.sf.jaer.eventio.ros.RosbagFileInputStream getNextRawPacket
WARNING: Discarding event with nonmonotonic timestamp detected for event 12102 in this message; delta time=-998966
Jan 31, 2018 8:35:16 AM net.sf.jaer.eventio.ros.RosbagFileInputStream getNextRawPacket
WARNING: Discarding event with nonmonotonic timestamp detected for event 9569 in this message; delta time=-998985
Jan 31, 2018 8:35:17 AM net.sf.jaer.eventio.ros.RosbagFileInputStream getNextRawPacket`

tobidelbruck, 2018-01-31:
I take back last observation. It appears the entire stream jumps back about 99ms, then continues from this point.

guillermogb, 2018-01-31:
Not answering the question about the jump backward in time, but related to the jAER rosbag reader....
There is a converter from ROS bag to Aedat in Sim's repository: 
https://github.com/inivation/AedatTools/blob/master/Matlab/ExampleConvertRosbagToAedat.m
I am not sure if you want to covert a few ros bags or if you want to have a very efficient converter to include in jAER, but the example might by just what you need for simple tests.

tobidelbruck, 2018-01-31:
Nice tip, thanks. Unfortunately it needs the matlab rosbag tools that only run on linux/mac os systems....
 https://github.com/bcharrow/matlab_rosbag
Bummer, but thanks anyhow. I guess I could do the conversion on a VM I have.

eliasm, 2018-02-02:
I have not observed this behavior on other sensors (DVS and DAVIS 240C). It seems peculiar that the delta is always around the same value (1s, not 99ms, if I read the numbers right). There might be a problem with the wrap-around. Could you maybe plot the event timestamps to get more insights about the behavior? How often does this happen (maybe once every second)?

tobidelbruck, 2018-02-05:
I believe I must be combining the timestamp ms and ns incorrectly. In java-bag-reader, (https://github.com/swri-robotics/bag-reader-java) the ms timestamp returned by getTime on Timestamp (https://docs.oracle.com/javase/8/docs/api/java/sql/Timestamp.html) already includes the ns when computing the ms. Therefore, to get the timestamp in us, I try to compensate for this by subtracting this ns part of the ms and then adding back the ns/1000 to the ms.... Hope this is clear. I cannot tell yet if my computation is correct. Here is my code to compute the relative AER timestamp in us from the start of the recording
```

 private boolean nonMonotonicTimestampDetected = false; // flag set by nonmonotonic timestamp if detection enabled

    private int getTimestamp(Timestamp timestamp) {
        long tsNs = timestamp.getNanos(); // gets the fractional seconds in ns
        long tsMs = timestamp.getTime() - (tsNs / 1000000); // this Timestamp ms includes the nanos already!, so subtract off the ns part
        long timestampUsAbsolute = tsMs * 1000+ tsNs / 1000; // ms*1000 =us and ns/1000=us, not sure about overflow however TODO check
        if (!firstTimestampWasRead) {
            firstTimestampUsAbsolute = timestampUsAbsolute;
            firstTimestampWasRead = true;
        }
        int ts = (int) (timestampUsAbsolute - firstTimestampUsAbsolute);
        final int dt = ts - mostRecentTimestamp;
        if (dt < 0 && nonMonotonicTimestampExceptionsChecked) {
            log.warning("Discarding event with nonmonotonic timestamp "+timestamp+" detected; delta time=" + dt);
            mostRecentTimestamp = ts;
            nonMonotonicTimestampDetected = true;
        } else {
            nonMonotonicTimestampDetected = false;
        }
        return ts;
    }

```