uzh-rpg/rpg_dvs_ros #58: IMU units for acceleration?

tobidelbruck, 2018-02-04:
I wonder if it has been checked that the units for the DAVIS IMU acceleration are really correct?  I am seeing in one recording about -0.9 m/s^2 for y component, which should be -9.8m/s^2 or about -1g.  This comes up in developing a native-java ROS bag reader for jaer, where I need to convert back from ros IMU units to internal jAER units, which are in g for acceleration and deg/s for gyro rotation (and are the native units for the invensense IMU, which is why these units were unfortunately used to begin with).

Does the ROS IMU acceleration have gravity component removed from y component? This wouldn't make too much sense but could it be why y is too small? It doesn't really make sense, since y component is always the one with significant negative component, and corresponds to how the camera was mounted in the recording. 
I haven't looked at the source code yet, sorry for this laziness.

tobidelbruck, 2018-02-05:
I see line 628 of https://github.com/uzh-rpg/rpg_dvs_ros/blob/master/davis_ros_driver/src/driver.cpp should convert from caer g units to m/s^2 unit. (I don't know what the switch is there for, since caer should deliver the IMU data in same form for each type of DAVIS).

```

                       if (davis_info_.chipID == DAVIS_CHIP_DAVIS346B)
                        {
                            // convert from g's to m/s^2 and align axes with camera frame
                            msg.linear_acceleration.x = -caerIMU6EventGetAccelY(event) * STANDARD_GRAVITY;
                            msg.linear_acceleration.y = -caerIMU6EventGetAccelX(event) * STANDARD_GRAVITY;
                            msg.linear_acceleration.z = -caerIMU6EventGetAccelZ(event) * STANDARD_GRAVITY;
                            // convert from deg/s to rad/s and align axes with camera frame
                            msg.angular_velocity.x = -caerIMU6EventGetGyroY(event) / 180.0 * M_PI;
                            msg.angular_velocity.y = -caerIMU6EventGetGyroX(event) / 180.0 * M_PI;
                            msg.angular_velocity.z = -caerIMU6EventGetGyroZ(event) / 180.0 * M_PI;
                        }
                        else
                        {
                            // convert from g's to m/s^2 and align axes with camera frame
                            msg.linear_acceleration.x = -caerIMU6EventGetAccelX(event) * STANDARD_GRAVITY;
                            msg.linear_acceleration.y = caerIMU6EventGetAccelY(event) * STANDARD_GRAVITY;
                            msg.linear_acceleration.z = -caerIMU6EventGetAccelZ(event) * STANDARD_GRAVITY;
                            // convert from deg/s to rad/s and align axes with camera frame
                            msg.angular_velocity.x = -caerIMU6EventGetGyroX(event) / 180.0 * M_PI;
                            msg.angular_velocity.y = caerIMU6EventGetGyroY(event) / 180.0 * M_PI;
                            msg.angular_velocity.z = -caerIMU6EventGetGyroZ(event) / 180.0 * M_PI;
                        }

```
Line 112 of https://github.com/uzh-rpg/rpg_dvs_ros/blob/master/davis_ros_driver/include/davis_ros_driver/driver.h defines 
`
static constexpr double STANDARD_GRAVITY = 9.81;

`

https://github.com/inilabs/libcaer/blob/master/include/events/imu6.h defines cAER units for acceleration as g's. Therefore I think that the g units of cAER are being multiplied by another factor of g in this code, rather than being divided by it. I must be missing something.

guillermogb, 2018-02-06:
Thanks for pointing this out. I think the switch is now obsolete. This code was there to fix an earlier version of libcaer regarding the DAVIS346B. 
https://github.com/uzh-rpg/rpg_dvs_ros/pull/17/commits/7e741c2b3cbc35458d5698a3428c2ca70ba0d027

It seems that libcaer has been fixed in the time since that commit, making sure that the IMU for the DAVIS346B has the same notation as the one for the DAVIS240C, and we did not notice. We did not fix it since we always compute the camera-to-IMU transformation (rotation and translation) using Kalibr.
We will remove this now obsolete switch.  @eliasm @supitalp just be aware.

guillermogb, 2018-02-06:
Fixed in PR https://github.com/uzh-rpg/rpg_dvs_ros/pull/59

closed 2018-02-06 by supitalp

tobidelbruck, 2018-02-06:
Thanks Guillermo, and reading the code I now I see I was mistaken. A 1g acceleration from cAER is correctly multiplied by 9.8m/s^2 to result in a final 9.8m/s^2 ROS value.
It still doesn't explain why I'm getting wrong values in my jAER rosbag acceleration, but it is probably something in that RosBagFileInputStream class, not here.

Have you validating physical correctness of the acceleration values by application using rpg-dvs code?

guillermogb, 2018-02-07:
Hi Tobi. We have validated the physical correctness of the acceleration values for the DAVIS240C sensors. In case of the mini DAVIS346B, we have reported to iniLabs that the values given by the accelerometer are off by some amount, and so, we have to calibrate the IMU (while at rest) and compensate for this offset whenever we use our visual-inertial odometry algorithms.