uzh-rpg/rpg_dvs_ros #60: Error when I run catkin build the "dvs_calibration package"

Merwanski, 2018-03-20:
I want to calibrate the dvs128 using the ROS-package but whenever I build the dvs_calibration package"
I keep getting the following error (I am using ubuntu 16 with ROS-kinetic)

$ catkin build dvs_calibration

[build] Found '9' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Starting  >>> catkin_simple                                                    
Finished  <<< catkin_simple                  [ 0.1 seconds ]                   
Starting  >>> dvs_msgs                                                         
Finished  <<< dvs_msgs                       [ 0.3 seconds ]                   
Starting  >>> dvs_calibration                                                  
_______________________________________________________________________________
Errors     << dvs_calibration:make /home/merwan/dvs_ros_ws/logs/dvs_calibration/build.make.000.log
/home/merwan/dvs_ros_ws/src/rpg_dvs_ros/dvs_calibration/src/circlesgrid.cpp:544:1: error: reference to ‘CirclesGridFinderParameters’ is ambiguous
 CirclesGridFinderParameters::CirclesGridFinderParameters()
 ^
In file included from /home/merwan/dvs_ros_ws/src/rpg_dvs_ros/dvs_calibration/src/circlesgrid.cpp:44:0:
/home/merwan/dvs_ros_ws/src/rpg_dvs_ros/dvs_calibration/include/dvs_calibration/circlesgrid.hpp:125:8: note: candidates are: struct CirclesGridFinderParameters
 struct CirclesGridFinderParameters
        ^
In file included from /opt/ros/kinetic/include/opencv-3.3.1/opencv2/calib3d/calib3d.hpp:48:0,
                 from /home/merwan/dvs_ros_ws/src/rpg_dvs_ros/dvs_calibration/include/dvs_calibration/circlesgrid.hpp:54,
                 from /home/merwan/dvs_ros_ws/src/rpg_dvs_ros/dvs_calibration/src/circlesgrid.cpp:44:
/opt/ros/kinetic/include/opencv-3.3.1/opencv2/calib3d.hpp:762:28: note:                 struct cv::CirclesGridFinderParameters
 struct CV_EXPORTS_W_SIMPLE CirclesGridFinderParameters
....
....
/opt/ros/kinetic/include/opencv-3.3.1/opencv2/calib3d.hpp:762:28: note:                 struct cv::CirclesGridFinderParameters
 struct CV_EXPORTS_W_SIMPLE CirclesGridFinderParameters
                            ^
make[2]: *** [CMakeFiles/camera_dvs.dir/src/circlesgrid.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/stereo.dir/src/circlesgrid.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/mono.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/stereo.dir/all] Error 2
make[1]: *** [CMakeFiles/camera_dvs.dir/all] Error 2
make: *** [all] Error 2
cd /home/merwan/dvs_ros_ws/build/dvs_calibration; catkin build --get-env dvs_calibration | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << dvs_calibration:make           [ Exited with code 2 ]            
Failed    <<< dvs_calibration                [ 14.5 seconds ]                  

How can I fix this issue ?

luca-della-vedova, 2018-09-17:
Hi,

I know it's been a few months but can you try out #72 if you didn't fix the problem already?

llongi, 2020-09-28:
Should have been fixed a long time ago.

closed 2020-09-28 by llongi