uzh-rpg/rpg_dvs_ros #63: libboost error on driver.cpp

smmmmi, 2018-05-06:
Hello,
On ubuntu 16.04 / ROS kinetic 
Following the build instruction
When building "catkin build dvs_ros_driver" I get errors with boost.
My boost version is the 1.58 default version installed with ROS.
same error also with "catkin build davis_ros_driver"
Any clues?
-------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/kinetic
Workspace:                   /home/mohammad/catkin_ws
-------------------------------------------------------------
Source Space:       [exists] /home/mohammad/catkin_ws/src
Log Space:          [exists] /home/mohammad/catkin_ws/logs
Build Space:        [exists] /home/mohammad/catkin_ws/build
Devel Space:        [exists] /home/mohammad/catkin_ws/devel
Install Space:      [unused] /home/mohammad/catkin_ws/install
DESTDIR:            [unused] None
-------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------
[build] Found '9' packages in 0.0 seconds.                                                                                 
[build] Package table is up to date.                                                                                       
Starting  >>> catkin_simple                                                                                                
Finished  <<< catkin_simple                 [ 0.1 seconds ]                                                                
Starting  >>> dvs_msgs                                                                                                     
Starting  >>> libcaer_catkin                                                                                               
Finished  <<< libcaer_catkin                [ 0.1 seconds ]                                                                
Finished  <<< dvs_msgs                      [ 0.1 seconds ]                                                                
Starting  >>> dvs_ros_driver                                                                                               
___________________________________________________________________________________________________________________________
Errors     << dvs_ros_driver:make /home/mohammad/catkin_ws/logs/dvs_ros_driver/build.make.007.log                          
/home/mohammad/catkin_ws/src/rpg_dvs_ros/dvs_ros_driver/src/driver.cpp: In constructor ‘dvs_ros_driver::DvsRosDriver::DvsRosDriver(ros::NodeHandle&, ros::NodeHandle)’:
/home/mohammad/catkin_ws/src/rpg_dvs_ros/dvs_ros_driver/src/driver.cpp:65:78: error: no matching function for call to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000000l>::subsecond_duration(double)’
   delta_ = boost::posix_time::microseconds(1e6/current_config_.streaming_rate);
                                                                              ^
/home/mohammad/catkin_ws/src/rpg_dvs_ros/dvs_ros_driver/src/driver.cpp:65:78: note: candidates are:
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/lock_types.hpp:18,
                 from /usr/local/include/boost/thread/pthread/thread_data.hpp:12,
                 from /usr/local/include/boost/thread/thread_only.hpp:17,
                 from /usr/local/include/boost/thread/thread.hpp:12,
                 from /usr/local/include/boost/thread.hpp:13,
                 from /home/mohammad/catkin_ws/src/rpg_dvs_ros/dvs_ros_driver/include/dvs_ros_driver/driver.h:22,
                 from /home/mohammad/catkin_ws/src/rpg_dvs_ros/dvs_ros_driver/src/driver.cpp:16:
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note: template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<Functor>, void>::type*)
     explicit subsecond_duration(T const& ss,
              ^
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note:   template argument deduction/substitution failed:
/usr/local/include/boost/date_time/time_duration.hpp: In substitution of ‘template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<Functor>, void>::type*) [with T = double]’:
/home/mohammad/catkin_ws/src/rpg_dvs_ros/dvs_ros_driver/src/driver.cpp:65:78:   required from here
/usr/local/include/boost/date_time/time_duration.hpp:285:14: error: no type named ‘type’ in ‘struct boost::enable_if<boost::is_integral<double>, void>’
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
[... 133 lines cut ...]
                 from /usr/local/include/boost/thread.hpp:13,
                 from /home/mohammad/catkin_ws/src/rpg_dvs_ros/dvs_ros_driver/include/dvs_ros_driver/driver.h:22,
                 from /home/mohammad/catkin_ws/src/rpg_dvs_ros/dvs_ros_driver/src/driver.cpp:16:
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000000l>::subsecond_duration(const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000000l>&)
   class BOOST_SYMBOL_VISIBLE subsecond_duration : public base_duration
                              ^
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note:   no known conversion for argument 1 from ‘double’ to ‘const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000000l>&’
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000000l>::subsecond_duration(boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000000l>&&)
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note:   no known conversion for argument 1 from ‘double’ to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000000l>&&’
make[2]: *** [CMakeFiles/dvs_ros_driver_nodelet.dir/src/driver.cpp.o] Error 1
make[1]: *** [CMakeFiles/dvs_ros_driver_nodelet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/dvs_ros_driver.dir/src/driver.cpp.o] Error 1
make[1]: *** [CMakeFiles/dvs_ros_driver.dir/all] Error 2
make: *** [all] Error 2
cd /home/mohammad/catkin_ws/build/dvs_ros_driver; catkin build --get-env dvs_ros_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................
Failed     << dvs_ros_driver:make           [ Exited with code 2 ]                                                         
Failed    <<< dvs_ros_driver                [ 2.3 seconds ]                                                                
[build] Summary: 3 of 4 packages succeeded.                                                                                
[build]   Ignored:   5 packages were skipped or are blacklisted.                                                           
[build]   Warnings:  None.                                                                                                 
[build]   Abandoned: None.                                                                                                 
[build]   Failed:    1 packages failed.                                                                                    
[build] Runtime: 2.5 seconds total

byteofsoren, 2018-07-17:
Yea i have the same problem. https://pastebin.com/PXuGjNeE

llongi, 2020-09-11:
Fixed in PR #108.

closed 2020-09-11 by llongi