uzh-rpg/rpg_dvs_ros #65: Incomplete instructions

nelkabosal, 2018-06-12:
First, for novices, it would be good to initially reference http://wiki.ros.org/kinetic/Installation/Ubuntu to know how to "Setup your sources.list" and "Set up your keys", and I assume I need to "Initialize rosdep"

Furthermore, step 8 appears incomplete; more detailed instructions would greatly benefit novices like myself.

"Only a udev rule is needed to run the DVS driver. An installation script is provided in the package `libcaer_catkin`.
`$ roscd libcaer_catkin' (need to source your `setup.bash` file first, or just do` $ cd libcaer_catkin`)
`$ sudo ./install.sh`
"

What exactly do I need to do for the `udev` rule? When it references the installation script in `libcaer_catkin`, is that a prerequisite for `roscd`? I also got "No command 'roscd' found"; is `roscd` a typo? Which setup.bash file is sourced? Am I not supposed to worry about that if I do `cd ibcaer_catkin`? That doesn't seem to help. 
Bottom line, I'm confused. Any help/advice would be appreciated. Thank you.

iliis, 2018-06-12:
These questions are not really specific to this package, but rather are generic ROS (or even just Linux) questions, which is probably why these things are not explained there. I agree tough that it would help to have more detailed instructions to get up and running with ROS and the packages from RPG.

The whole instructions assume you have a working ROS installation on your system (prefereably Kinetic on Ubuntu). If you need help with that ask me or somebody else at the lab directly ;)

So, to actually answer your questions:
After you set up ROS you need to create a ROS workspace: Execute `catkin init` in the folder of choice. This folder should already contain a (possibly empty) `src` folder: (step 4)

```
cd wherever/you/want/
mkdir my_ros_ws
cd my_ros_ws
mkdir src
catkin config --init --mkdirs --extend /opt/ros/kinetic --merge-devel --cmake-args -DCMAKE_BUILD_TYPE=Release
```
(Note that there are different tools to do these things. The most current is catkin-tools, see http://catkin-tools.readthedocs.io/en/latest/ You shouldn't need to use rosbuild or catkin_make, even though there are a lot of old instructions on the web using those.)
You now should have working catkin workspace. However, step 4 fails to mention that you need to `source setup.bash` after this, to actually use the workspace. In your workspace, there should now be a `devel` folder, so you can execute: `source devel/setup.bash`. This will "load" your workspace (by defining some commands, loading paths etc.). You should now be able to call things like `roscd` or `roslaunch`. If you forgot to source the setup.bash file you will get the 'no command "roscd"' error.

For the udev rule all you need to do is copy the file `rpg_dvs_ros/libcaer_catkin/65-inilabs.rules` to `/etc/udev/rules.d/`. The `install.sh` script does just that ;) So you can either `cd` manually to the correct dir and copy the udev rule-file by yourself or use `roscd` to directly jump to the package folder and run the script which does nothing else than copying the file ;)

nelkabosal, 2018-06-12:
Thank you for the feedback. That helped me get the basic examples to run.

closed 2018-07-17 by magehrig