uzh-rpg/rpg_dvs_ros #68: undefined reference to `libusb_strerror'

CaptainJL, 2018-08-13:
I am having a compile error when building 'davis_ros_driver'

`/media/captainjl/DATA/catkin_ws_dvs/devel/lib/libcaer.so: undefined reference to `libusb_strerror'`

Does anyone know how to solve this? 
I have installed the libusb as defined in the readme file.

fabien-colonnier, 2018-11-22:
The same problem appeared to me after installing a new node in my workspace. I solved it by adding `usb-1.0` in the target_link_libraries of the CMakeLists.txt file, as follows:
`target_link_libraries(dvs_ros_driver
  ${catkin_LIBRARIES}
  ${Boost_LIBRARIES}
  usb-1.0
)`

As it shouldn't be necessary, I still think another way should be possible...

llongi, 2020-09-11:
Official libcaer build is now used, see README for further details.

closed 2020-09-11 by llongi