uzh-rpg/rpg_dvs_ros #70: eDVS support

jackokaiser, 2018-08-23:
Hi,

As far as I can tell, there is no ros wrapper for the [eDVS](https://inivation.com/support/hardware/edvs/). I was able to tweak the DVS128 driver to work with eDVS since libcaer also wraps the eDVS in a similar way than the DVS128 ([see the diff)](https://github.com/jackokaiser/rpg_dvs_ros/commit/ae155207565b454ccb5177ec903fbccad094fde4).

Would you be interested in a cleaned-up pull request, and if yes, what would be the best way to integrate the eDVS properly?

Cheers,
Jacques

guillermogb, 2018-08-27:
Hi Jacques,
I was checking which version of the driver did we internally use with the eDVS (I know we used it in one of our projects),... we had the code running directly on the eDVS, without an intermediate computer. It would be nice if you could make a clean, non-destructive pull request to support the eDVS.
The best way to integrate the eDVS properly...? is to make the eDVS publish events in the same format as the DVS and the DAVIS, so that these can be read and processed in the same way, regardless of the source, if possible.
Best,
Guillermo

jackokaiser, 2019-05-15:
Hi Guillermo,

Since I was using with the pushbot, I ended up integrating ROS and the event format in the pushbot driver from Terry: https://github.com/jackokaiser/nstbot

This makes more sense since the robot also has other sensors that should be read.

Cheers,
Jacques

guillermogb, 2019-05-15:
Thanks for letting us know.
Cheers,
Guillermo

closed 2019-05-15 by guillermogb