uzh-rpg/rpg_dvs_ros #79: TimeStamp Reset Problem.

PanicHappyDeaf, 2018-11-03:
Hi,

After I using rosbag record to recording my Davis data, I used dvs_write_file( extract_davis_bag.py) to obtain internal data. However, I notice that time stamps in the file event.txt did not be reset to 0.0000. Is there something wrong with my operation or some details have been missed through my process?

BW,

Thank you.

guillermogb, 2018-11-04:
Hi. I think this is normal. If you want the timestamps to start at zero you have to subtract the timestamp of the first event from all of them. Regards.

guillermogb, 2018-11-04:
True, the reset_timestamp option is available in the python script. I just run a quick test and it seems to work fine. I used the `slider_depth.bag` from the [Event Camera Dataset](http://rpg.ifi.uzh.ch/davis_data.html)

Command:
```
python extract_davis_bag_files.py path_to_file/slider_depth.bag  --reset_time
```
Output:
```
To extract depth image of simulated scenes, please install OpenEXR and Imath
Reset time: 1461581614.988127335
All data extracted!
Events:       1078541
Images:       87
Depth maps:   0
IMU:          0
Ground truth: 339
Removed IMU file since there were no messages.
Removed depthmaps file since there were no messages.
```
It created several files, including `events.txt`, whose first ten lines are: 
```
0.003811000 96 133 0
0.003820001 127 171 0
0.003836000 4 160 0
0.003837000 149 122 0
0.003848001 63 121 1
0.003849001 17 144 1
0.003852000 92 119 0
0.003866001 16 137 1
0.003875000 156 71 0
0.003879000 26 149 0
```
so they indeed have the timestamp offset (1461581614.988127335) removed.

PanicHappyDeaf, 2018-11-04:
Sorry for taking you too much time. Thank you for your help, I really appreciated that.

guillermogb, 2018-11-04:
No worries

closed 2018-11-04 by guillermogb