uzh-rpg/rpg_dvs_ros #87: where can i find dvs_ros_driver::DVS_ROS_DriverConfig?

eugenelyj, 2019-04-30:
in rpg_dvs_ros/dvs_ros_driver/include/dvs_ros_driver/driver.h, it include `<dvs_ros_driver/DVS_ROS_DriverConfig.h>`, which i can not find in source code...where can i get it?

guillermogb, 2019-04-30:
I have that file in my devel folder:
`catkin_ws/devel/include/dvs_ros_driver/DVS_ROS_DriverConfig.h`
so I assume that it is automatically generated by the `catkin build` command.

You may find the file on Linux by typing the commands:
    `sudo updatedb`
and then
    `locate DVS_ROS_DriverConfig`

eugenelyj, 2019-05-01:
> I have that file in my devel folder:
> `catkin_ws/devel/include/dvs_ros_driver/DVS_ROS_DriverConfig.h`
> so I assume that it is automatically generated by the `catkin build` command.
> 
> You may find the file on Linux by typing the commands:
> `sudo updatedb`
> and then
> `locate DVS_ROS_DriverConfig`

eugenelyj, 2019-05-01:
> I have that file in my devel folder:
> `catkin_ws/devel/include/dvs_ros_driver/DVS_ROS_DriverConfig.h`
> so I assume that it is automatically generated by the `catkin build` command.
> 
> You may find the file on Linux by typing the commands:
> `sudo updatedb`
> and then
> `locate DVS_ROS_DriverConfig`

Yes, it is auto-generated by dynamic_reconfigure package. Thanks :)

closed 2019-05-01 by eugenelyj