uzh-rpg/rpg_dvs_ros #94: change streaming rate

plutozyw, 2020-01-09:
hi, i use GUI to modify the parameters .
No matter how I change the streaming rate, the actual frequency will not exceed 100 HZ

rostopic hz /dvs_rendering
subscribed to [/dvs_rendering]
average rate: 99.925
	min: 0.000s max: 0.022s std dev: 0.00625s window: 95
average rate: 99.971
	min: 0.000s max: 0.022s std dev: 0.00671s window: 195

How can I make the frequency greater than 100hz, thanks!

JuSquare, 2021-04-19:
Hi @plutozyw 

Did you find a way to update the streaming rate? Also, have you noticed that it is not really stable? In my case the rate never exceeds 100Hz as for you, and it changes from time to time. 

Thanks!

llongi, 2021-08-19:
Hello, this is probably because the ROS driver uses the libcaer library to talk to iniVation cameras, and by default it delivers data in 10ms intervals, resulting in a max 100Hz frequency. This can be changed by editing the ROS driver code and recompiling. In 'davis_ros_driver/src/driver.cpp', go to line 633 and after the line:

`caerDeviceConfigSet(davis_handle_, CAER_HOST_CONFIG_DATAEXCHANGE, CAER_HOST_CONFIG_DATAEXCHANGE_BLOCKING, true);`

add the following line:
`
caerDeviceConfigSet(davis_handle_, CAER_HOST_CONFIG_PACKETS, CAER_HOST_CONFIG_PACKETS_MAX_CONTAINER_INTERVAL, 1000);`
​
recompile and install. This will lower the maximum time interval per data packet to 1ms (1000µs), which should allow you to reach a 1000 Hz rate in ROS. This will probably have a negative impact on performance!

closed 2021-08-19 by llongi