uzh-rpg/rpg_dvs_ros #110: Unable to install libcaer

BonJovi1, 2020-09-22:
Hi,
Thank you for the wonderful ROS DVS packages! 
I'm using ROS Kinetic on Ubuntu 16.04.7. `libusb` is installed, but I'm unable to install `libcaer`. According to the latest readme, I'm running, 
```
sudo apt-get install libcaer-dev
```
But I get an error saying,
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libcaer-dev
```
I looked up the official iniVation documentation for installing `libcaer`. it asked me to do two extra steps:
```
$ sudo add-apt-repository ppa:inivation-ppa/inivation
$ sudo apt-get update
$ sudo apt-get install libcaer-dev
```
I did this too, but it still gives the same error. Could you kindly advise me on what might be going wrong?
Thank you,
Abhinav

llongi, 2020-09-25:
Please check the iniVation instructions again, for Ubuntu Xenial there is limited, separate support. Reading further "For Ubuntu Xenial (16.04 LTS), we offer a separate PPA repository supporting the x86_64 and x86 architectures.", instructions are available then on how to install this.

closed 2020-09-25 by llongi

yuguanfeng, 2021-04-20:
Only adding one PPA repository is not enough and it still gives the same error. But after I add all the four PPA repositories from the iniVation instructions, it works!

llongi, 2021-07-27:
Please see the Ubuntu Xenial instructions (last paragraph) here: https://inivation.gitlab.io/dv/dv-docs/docs/getting-started/#ubuntu-linux
Please also note that support for Xenial is discontinued and will receive no more updates.

li10141110, 2021-07-27:
Thank you so much for your response!

ramy-bat, 2022-01-06:
updated#36