uzh-rpg/rpg_dvs_ros #112: grayscale mode image_raw davis346  becomes binary image

HiOtto, 2020-12-04:
The one with problem
[image]
The normal one
[image]

SO why it becomes binary image ? The setting with davis_ros_driver are the same:

{DiffBn_coarse: 4, DiffBn_fine: 39, OFFBn_coarse: 4, OFFBn_fine: 0, ONBn_coarse: 6,
  ONBn_fine: 200, PrBp_coarse: 2, PrBp_fine: 58, PrSFBp_coarse: 1, PrSFBp_fine: 33,
  RefrBp_coarse: 4, RefrBp_fine: 25, aps_enabled: true, dvs_enabled: false, exposure: 6000,
  frame_interval: 40000,
  imu_acc_scale: 3, imu_enabled: true, imu_gyro_scale: 3, max_events: 0,
  streaming_rate: 30}

llongi, 2020-12-19:
Do both camer's outputs look fine in the DV SDK?
Did you disable AutoExposure and really set the Exposure to the same value on both?

HiOtto, 2020-12-29:
> disable AutoExposure and really set the Exposure to the same value on both?

yes, the same config file for the 2 cameras as below