uzh-rpg/rpg_dvs_ros #117: Connection to node=/davis_ros_driver failed: param client failed

Bochen1997, 2021-09-14:
Dear Developers,

Following the steps in the webpage, I have installed this driver. The system version is Ubuntu 18.04 LTS.

However, I found that there are some errors when I use "roslaunch dvs_renderer davis_mono.launch" with a DAVIS 240C camera. The driver cannot connect the DAVIS 240C, as shown in the following log.

---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
y1412@labComputer:~/catkin_ws$ catkin build dvs_renderer
----------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/y1412/catkin_ws
----------------------------------------------------------
Build Space:        [exists] /home/y1412/catkin_ws/build
Devel Space:        [exists] /home/y1412/catkin_ws/devel
Install Space:      [unused] /home/y1412/catkin_ws/install
Log Space:          [exists] /home/y1412/catkin_ws/logs
Source Space:       [exists] /home/y1412/catkin_ws/src
DESTDIR:            [unused] None
----------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
----------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------
[build] Found '9' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> catkin_simple                                                    
Finished  <<< catkin_simple                [ 0.2 seconds ]                     
Starting  >>> dvs_msgs                                                         
Finished  <<< dvs_msgs                     [ 0.2 seconds ]                     
Starting  >>> dvs_renderer                                                     
Finished  <<< dvs_renderer                 [ 0.3 seconds ]                     
[build] Summary: All 3 packages succeeded!                                     
[build]   Ignored:   6 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 0.8 seconds total.                                            
y1412@labComputer:~/catkin_ws$ V
V: command not found
y1412@labComputer:~/catkin_ws$ source ~/catkin_ws/devel/setup.bash
y1412@labComputer:~/catkin_ws$ roslaunch dvs_renderer davis_mono.launch
... logging to /home/y1412/.ros/log/675cb8b4-1549-11ec-92e6-d8cb8af4246b/roslaunch-labComputer-4052.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://labComputer:43127/

SUMMARY
========

PARAMETERS
 * /dvs_renderer/display_method: red-blue
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    davis_ros_driver (davis_ros_driver/davis_ros_driver)
    dvs_renderer (dvs_renderer/dvs_renderer)
    image_view (rqt_image_view/rqt_image_view)
    rqt_reconfigure (rqt_reconfigure/rqt_reconfigure)

auto-starting new master
process[master]: started with pid [4063]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 675cb8b4-1549-11ec-92e6-d8cb8af4246b
process[rosout-1]: started with pid [4074]
started core service [/rosout]
process[davis_ros_driver-2]: started with pid [4077]
process[dvs_renderer-3]: started with pid [4078]
process[image_view-4]: started with pid [4079]
process[rqt_reconfigure-5]: started with pid [4080]
[ WARN] [1631616556.392689040]: Camera calibration file /home/y1412/.ros/camera_info/DAVIS-08360036.yaml not found.
[davis_ros_driver-2] process has died [pid 4077, exit code -11, cmd /home/y1412/catkin_ws/devel/lib/davis_ros_driver/davis_ros_driver __name:=davis_ros_driver __log:=/home/y1412/.ros/log/675cb8b4-1549-11ec-92e6-d8cb8af4246b/davis_ros_driver-2.log].
log file: /home/y1412/.ros/log/675cb8b4-1549-11ec-92e6-d8cb8af4246b/davis_ros_driver-2*.log
[WARN] [1631616561.950153]: wait_for_service(/davis_ros_driver/set_parameters): failed to contact, will keep trying
[ERROR] [1631616565.027345]: Connection to node=/davis_ros_driver failed:
param client failed
Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_reconfigure/treenode_qstditem.py", line 73, in run
    self._raw_param_name
ROSException: timeout exceeded while waiting for service /davis_ros_driver/set_parametersTreenodeQstdItem. Couldn't connect to /davis_ros_driver
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Could you give me some advice for this case? Thanks a lot for your help.

ShifanZhu, 2021-09-14:
Have a similar situation. Below is the gdb information I got.

========

$ gdb ./davis_ros_driver core
GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./davis_ros_driver...(no debugging symbols found)...done.
/home/zh/code/uslam_ws/devel/lib/davis_ros_driver/core: No such file or directory.
(gdb) r
Starting program: /home/zh/code/uslam_ws/devel/lib/davis_ros_driver/davis_ros_driver 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffed10d700 (LWP 8467)]
[New Thread 0x7fffec90c700 (LWP 8468)]
[New Thread 0x7fffe7fff700 (LWP 8469)]
[New Thread 0x7fffe77fe700 (LWP 8470)]
[New Thread 0x7fffe6ffd700 (LWP 8471)]
[Thread 0x7fffe6ffd700 (LWP 8471) exited]
[New Thread 0x7fffe6ffd700 (LWP 8472)]
[New Thread 0x7fffe67fc700 (LWP 8474)]
[ INFO] [1631639895.751392305]: DAVIS ID-1 SN-00000545 [2:2] --- ID: 1, Master: 1, DVS X: 346, DVS Y: 260, Logic: 6.

[New Thread 0x7fffe5ffb700 (LWP 8475)]
[New Thread 0x7fffe57fa700 (LWP 8476)]
[ INFO] [1631639896.082391450]: using default calibration URL
[ INFO] [1631639896.082514193]: camera calibration URL: file:///home/zh/.ros/camera_info/DAVIS-00000545.yaml
[ INFO] [1631639896.082627295]: Unable to open camera calibration file [/home/zh/.ros/camera_info/DAVIS-00000545.yaml]
[ WARN] [1631639896.082684750]: Camera calibration file /home/zh/.ros/camera_info/DAVIS-00000545.yaml not found.

Thread 10 "davis_ros_drive" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe57fa700 (LWP 8476)]
0x00007ffff58dede3 in std::_Rb_tree_decrement(std::_Rb_tree_node_base*) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff58dede3 in std::_Rb_tree_decrement(std::_Rb_tree_node_base*) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff6dac69c in boost::detail::add_new_tss_node(void const*, boost::shared_ptr<boost::detail::tss_cleanup_function>, void*) () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#2  0x00007ffff6dad5cc in boost::detail::set_tss_data(void const*, boost::shared_ptr<boost::detail::tss_cleanup_function>, void*, bool) () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#3  0x00007ffff77013f6 in ros::CallbackQueue::setupTLS() () from /opt/ros/melodic/lib/libroscpp.so
#4  0x00007ffff770428f in ros::CallbackQueue::callAvailable(ros::WallDuration) ()
   from /opt/ros/melodic/lib/libroscpp.so
#5  0x00007ffff774467e in ros::spinOnce() () from /opt/ros/melodic/lib/libroscpp.so
#6  0x00005555555910f1 in davis_ros_driver::DavisRosDriver::readout() ()
#7  0x00007ffff6dacbcd in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#8  0x00007ffff6b836db in start_thread (arg=0x7fffe57fa700) at pthread_create.c:463
#9  0x00007ffff5337a3f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb)

grammers, 2021-09-24:
I also have the some issue. I am getting a segmentation fault when I try to run davis_ros_driver by it self.
```
spotnuc@spotnuc-NUC10i5FNK:~/ros_workspaces/event_dvs/src/rpg_dvs_ros$ rosrun davis_ros_driver davis_ros_driver 
[ INFO] [1632478054.260927865]: DAVIS ID-1 SN-00000305 [2:7] --- ID: 1, Master: 1, DVS X: 346, DVS Y: 260, Logic: 6.

[ INFO] [1632478054.588147418]: using default calibration URL
[ INFO] [1632478054.588257331]: camera calibration URL: file:///home/spotnuc/.ros/camera_info/DAVIS-00000305.yaml
Segmentation fault (core dumped)
```

By disconnect an connecting the camera am I guessing the crash happens when the driver connects to the camera.

Also I successfully installed it and run it in June on a different pc and re-installation on that one works fine.

A wild guess is that it is a dependency that has updated and it causes the crash but I have no idea what dependency and it could be some thing else.

All this are done on ubuntu 18.04.

llongi, 2021-09-24:
Hello, the issue is most probably the new version of Boost we have in our iniVation PPA repository, a package called 'boost-inivation'. The ROS build system seems to get confused about which Boost to use at compile and runtime, which leads to this crash.
A quick fix is to uninstall the 'boost-inivation' package, then clean and rebuild the ROS packages, then they should work again.

Bochen1997, 2021-09-26:
> Hello, the issue is most probably the new version of Boost we have in our iniVation PPA repository, a package called 'boost-inivation'. The ROS build system seems to get confused about which Boost [...]

Thanks a lot for your help! :+1: :+1: :+1:

Bochen1997, 2021-09-26:
> I also have the some issue. I am getting a segmentation fault when I try to run davis_ros_driver by it self.
>
> ```
> spotnuc@spotnuc-NUC10i5FNK:~/ros_workspaces/event_dvs/src/rpg_dvs_ros$ rosrun [...]

@grammers Thanks for your good advice!

grammers, 2021-09-28:
By running `apt purge boost-inivation` I got it to work!
Thanks for the help!

closed 2021-11-25 by Bochen1997

llongi, 2021-12-13:
Just for your info, we're also fixing this on boost-inivation package side to prevent conflicts in the future.
A future release of boost-inivation will move the custom install to /opt/inivation/boost/ instead of /opt/boost-inivation/, which stops it from being auto-discovered and interfering with libcaer/ROS. I'll update this issue again when we release that update.

liuhuixiang-987, 2023-03-31:
> 仅供您参考，我们还在提升升级包方面修复了此问题，以防止将来发生冲突。 boost-inivation 的未来版本会将自定义安装移动到 /opt/inivation/boost/ 而不是 /opt/boost-inivation/，这会阻止它被自动发现并干扰 libcaer/ROS。当我们发布该更新时，我将再次更新此问题。

When I run sudo apt-get remove boost-inivation or sudo apt purge boost-inivation, I get this problem: Unable to locate package boost-inivation. I have updated the software source through the command "sudo apt-get update", but it has no effect. When I check whether the software has been installed through the command "sudo dpkg -l | grep*", there is no display, indicating that my system does not have this software package. So how to solve this problem "[WARN] [1680231915.210308595]: Camera calibration file /home/syou/.ros/camera_info/DAVIS-00000567.yaml not found."