uzh-rpg/rpg_dvs_ros #120: Could not find a package configuration file provided by "libcaer-dev" with   any of the following names:

wujihua628, 2021-11-30:
**When I run the code :catkin build davis_ros_driver, I got the problem**

> Errors     << davis_ros_driver:cmake /home/wjh/catkin_ws/logs/davis_ros_driver/build.cmake.010.log  
CMake Error at /home/wjh/catkin_ws/src/rpg_dvs_ros/davis_ros_driver/CMakeLists.txt:5 (find_package):
  By not providing "Findlibcaer-dev.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "libcaer-dev", but CMake did not find one.

  **Could not find a package configuration file provided by "libcaer-dev" with
  any of the following names:

    libcaer-devConfig.cmake
    libcaer-dev-config.cmake**

  Add the installation prefix of "libcaer-dev" to CMAKE_PREFIX_PATH or set
  "libcaer-dev_DIR" to a directory containing one of the above files.  If
  "libcaer-dev" provides a separate development package or SDK, be sure it
  has been installed.

I use melodic， davis, 

What's more, I have installed "libcaer-dev" successfully by the code "sudo apt-get install libcaer-dev", so I don't understand why this error occurred.

Hope you guys can help me.

best wishes，
xiaowu

wujihua628, 2021-12-01:
I have solved this problem!

According to the error message：
Add the installation prefix of "libcaer" to CMAKE_PREFIX_PATH or set
  "libcaer_DIR" to a directory containing one of the above files.  If
  "libcaer" provides a separate development package or SDK, be sure it has
  been installed.

I added this command to the CMakeLists.txt file：
set(libcaer_DIR "/usr/lib/x86_64-linux-gnu/cmake/libcaer/")
find_package(libcaer REQUIRED)
（The first line is the address of my llibcaerConfig.cmake file）

llongi, 2021-12-01:
Glad you solved it. Though that should not be required if the proper CMake paths are set up, /usr/lib/x86_64-linux-gnu/cmake/ is the default path for searching.
Could you try the following please:
- remove set(libcaer_DIR ...) line
- clear CMake cache (remove CMakeCache.txt file)
- check that it fails again
- go to /usr/lib/x86_64-linux-gnu/cmake/libcaer/ and rename 'libcaerConfig.cmake' to 'libcaer-config.cmake' and rename 'libcaerConfigVersion.cmake' to 'libcaer-config-version.cmake'
- try again, does it work now?

wujihua628, 2021-12-02:
Thank you for your reply，
I did as you told me, but still the same error as I did before：

```
Could not find a package configuration file provided by "libcaer" with any
  of the following names:

    libcaerConfig.cmake
    libcaer-config.cmake

```

llongi, 2021-12-02:
Okay, turns out that you have to pass _-DCMAKE_INSTALL_PREFIX=/usr_ for it to search in the right place, might be something changed in CMake.
Try the following catkin command, adapted from README:
`$ catkin config --init --mkdirs --extend /opt/ros/kinetic --merge-devel --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr`

wujihua628, 2021-12-02:
I tried the code :
``wjh@wjh:~/catkin_ws$ catkin config --init --mkdirs --extend /opt/ros/melodic --merge-devel --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr`
`
But it still had the same error
I still added the path to the CMakeLists.txt file to run successfully, but after I solved the problem, a new problem appeared：

```
CMake Error at /opt/ros/melodic/share/roscpp/cmake/roscppConfig.cmake:173 (message):
  Project 'davis_ros_driver' tried to find library 'pthread'.  The library is
  neither a target nor built/installed properly.  Did you compile project
  'roscpp'? Did you find_package() it before the subdirectory containing its
  code is included?
```

closed 2021-12-08 by wujihua628

llongi, 2021-12-13:
I see you're using ROS Kinetic, which is based on Ubuntu 16.04. We don't really support that anymore on our (iniVation's) side, so can't guarantee that libcaer works perfectly fine with it, it isn't tested. I guess some weird interaction with its very old CMake version is happening here that prevents automatic discovery of the CMake config file.
In any case, using `set(libcaer_DIR "/usr/lib/x86_64-linux-gnu/cmake/libcaer/")` can be a valid solution there.

wujihua628, 2021-12-14:
Thank you for your reply.
I use the Ubuntu18.04 Melodic. 
Because I had no idea to solve the problems, I reinstalled the Ubuntu 18.04 again. And I successfully recompiled it.