uzh-rpg/rpg_dvs_ros #121: file "libcaer.so.3.3.11" does not exist

friedhelm-h, 2021-11-30:
# Problem
Building the drivers of the Davis and the Dvxplorer with the command ```catkin build davis_ros_driver``` or ```catkin build dvxplorer_ros_driver``` respectively fails with the error:
```shell
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/libcaer/libcaer-exports.cmake:97 (message):
  The imported target "libcaer::caer" references the file

     "/usr/lib/x86_64-linux-gnu/libcaer.so.3.3.11"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/lib/x86_64-linux-gnu/cmake/libcaer/libcaer-exports.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/libcaer/libcaerConfig.cmake:58 (INCLUDE)
  CMakeLists.txt:5 (find_package)
```

The build worked before on the same machine but stopped working. A new installation of libcaer did not resolve the issue

# Environment
OS: Ubuntu 20.04
CMake: 3.16.3
catkin_tools: 0.8.0

guillermogb, 2021-11-30:
Dear @llongi , there seem to be two recent issues related to libcaer (https://github.com/uzh-rpg/rpg_dvs_ros/issues/120 and https://github.com/uzh-rpg/rpg_dvs_ros/issues/121). Could you please take a look?

llongi, 2021-12-01:
This looks like either a broken install or stale cache.
First make sure that both libcaer and libcaer-dev are version 3.3.12.
Then delete any catkin/CMake cache (like the CMakeCache.txt file) and try again.

closed 2021-12-01 by llongi