uzh-rpg/rpg_dvs_ros #122: driver does not work for DVXplorer Mini

phxhuy, 2021-12-02:
I have tried the ROS driver with a DVXplorer Mini camera (https://shop.inivation.com/collections/dvxplorer-mini) with the dvxplorer launch file ( dvxplorer_mono.launch), but it did not work. Here is the error I get: 

`[dvxplorer_ros_driver-2] process has died [pid 15576, exit code -11, cmd /home/huy/event_dvx_ws/devel/lib/dvxplorer_ros_driver/dvxplorer_ros_driver __name:=dvxplorer_ros_driver __log:=/home/huy/.ros/log/610fcdc0-5365-11ec-837e-380025620174/dvxplorer_ros_driver-2.log].`

I have installed the libcaer-dev library and build this package without any issue. I have also checked the camera with the DV software from Inivation, and the camera is working fine (pictured).
[image]

Could you please give me some suggestion to make it work? Thank you.

CharlChc, 2022-01-22:
I got the same problem with same error notice for my DVXplorer-Lite. Does anyone know how to solve this? Thanks in advance.

shrutarv, 2022-12-20:
Hi,

Were you able to run ultimate slam with dvxplorer?