uzh-rpg/rpg_dvs_ros #130: [ WARN] [1641968634.644978988]: Camera calibration file /home/zfl/.ros/camera_info/DAVIS-00000554.yaml not found.

987428377, 2022-01-12:
Dear master, I encountered the same problem as the web page https://giters.com/uzh-rpg/rpg_dvs_ros/issues/117.
But I find that  /opt/inivation/boost/ has been instead of /opt/boost-inivation/. The erroe is still displayed as follows:

 Camera calibration file /home/zfl/.ros/camera_info/DAVIS-00000554.yaml not found.

And I try uninstall boost-inivation, then I can't connnect to zhe DAVIS 346.

Can you help me ?  All this are done on ubuntu 18.04.

closed 2022-01-13 by 987428377

OnurSelim, 2022-11-04:
Hi, I am also getting the same error. I have tried the solution given in the web page but it did not work.

Is there anything that I can try? I am working on Ubuntu 20.04 by the way.

ZhdsGithub, 2023-03-10:
Actually I met the same problem, that is because you do not have a well-formed YAML file, all you need is to fix your calibration result into a suitable form like this.
///
image_width: 346
image_height: 260
camera_name: DAVIS
camera_matrix:
  rows: 3
  cols: 3
  data: [274.6631, 0, 168.47, 0, 274.59, 141.8423, 0, 0, 1]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.4043, 0.2296, 0.0003377, 0.000391, -0.080294]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1, 0, 0, 0, 1, 0, 0, 0, 1]
projection_matrix:
  rows: 3
  cols: 4
  data: [1, 0, 1, 0, 0, 1, 1, 0, 0, 0, 1, 0] 
///
Do not forget to move it to the right path like "/home/xxx/.ros/camera_info" by the way.

ZhdsGithub, 2023-03-10:
> Hi, I am also getting the same error. I have tried the solution given in the web page but it did not work.
> 
> Is there anything that I can try? I am working on Ubuntu 20.04 by the way.

Actually I met the same problem, that is because you do not have a well-formed YAML file, all you need is to fix your calibration result into a suitable form like this.
///
image_width: 346
image_height: 260
camera_name: DAVIS
camera_matrix:
  rows: 3
  cols: 3
  data: [274.6631, 0, 168.47, 0, 274.59, 141.8423, 0, 0, 1]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.4043, 0.2296, 0.0003377, 0.000391, -0.080294]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1, 0, 0, 0, 1, 0, 0, 0, 1]
projection_matrix:
  rows: 3
  cols: 4
  data: [1, 0, 1, 0, 0, 1, 1, 0, 0, 0, 1, 0] 
///
Do not forget to move it to the right path like "/home/xxx/.ros/camera_info" by the way.