uzh-rpg/rpg_dvs_ros #132: ERROR: cannot launch node of type [rqt_image_view/rqt_image_view]: rqt_image_view

wujihua628, 2022-01-18:
Hi,
I opened the rpg-ros-dvs package today and ran 

> roslaunch dvs_renderer davis_mono.launch

 but I don’t know why it went wrong. There was no error when I ran this package before, but today I ran into an error, which is so weird！
[image]

Any suggestion will help me，thanks！

2565583595long, 2024-02-22:
```bash
apt-cache policy rqt_image_view
apt-cache search dv-processing
sudo apt install ros-melodic-rqt-image-view
```

Weihua-Wang, 2024-07-24:
`sudo apt-get install ros-noetic-rqt
sudo apt-get install ros-noetic-rqt-common-plugins`

Weihua-Wang, 2024-07-24:
run
`sudo apt-get install ros-noetic-rqt`
`sudo apt-get install ros-noetic-rqt-common-plugins`