uzh-rpg/rpg_dvs_ros #133: roslaunch error:[dvxplorer_ros_driver-2] process has died [pid 10699, exit code -11

CharlChc, 2022-01-23:
Hi guys,
When I run the following command with my DVSploer-Lite plugged in,
**$roslaunch dvs_renderer dvxplorer_mono.launch**
I got error notice:
[dvxplorer_ros_driver-2] process has died [pid 10699, exit code -11, cmd /home/USERNAME/catkin_ws/devel/lib/dvxplorer_ros_driver/dvxplorer_ros_driver __name:=dvxplorer_ros_driver __log:=/home/USERNAME/.ros/log/6a4bef1a-7b63-11ec-bbcf-6c02e04089e4/dvxplorer_ros_driver-2.log].
log file: /home/USERNAME/.ros/log/6a4bef1a-7b63-11ec-bbcf-6c02e04089e4/dvxplorer_ros_driver-2*.log

Dose anyone come across this problem before? Thanks in advance.

CharlChc, 2022-01-24:
Hi Sir @llongi, may I have you help here?

juliuserbach, 2022-01-28:
I had the same issue with the DXplorer with Ubuntu 18.04 and ROS Melodic.
I removed libcaer and libcaer-dev and installed version 3.3.6 from source (https://github.com/inivation/libcaer/tree/3.3.6)
I hope this helps.