uzh-rpg/rpg_dvs_ros #134: DVS 346 stereo recording parallel both at the same timestamp

Darshanbhandarics, 2022-02-16:
I have connected two dvs 346 to two different usb port on my laptop running on ubuntu 18.04. 
when I run the command "roslaunch dvs_renderer stereo_davis.launch"
I am getting the following error:

roslaunch dvs_renderer stereo_davis.launch 
... logging to /home/lenovo/.ros/log/ef044dcc-8eed-11ec-afee-4c796e50746d/roslaunch-lenovo-4614.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lenovo:42741/

SUMMARY
========

PARAMETERS
 * /davis_left/davis_ros_driver/DiffBn_coarse: 4
 * /davis_left/davis_ros_driver/DiffBn_fine: 39
 * /davis_left/davis_ros_driver/OFFBn_coarse: 4
 * /davis_left/davis_ros_driver/OFFBn_fine: 0
 * /davis_left/davis_ros_driver/ONBn_coarse: 6
 * /davis_left/davis_ros_driver/ONBn_fine: 200
 * /davis_left/davis_ros_driver/PrBp_coarse: 2
 * /davis_left/davis_ros_driver/PrBp_fine: 58
 * /davis_left/davis_ros_driver/PrSFBp_coarse: 1
 * /davis_left/davis_ros_driver/PrSFBp_fine: 33
 * /davis_left/davis_ros_driver/RefrBp_coarse: 4
 * /davis_left/davis_ros_driver/RefrBp_fine: 25
 * /davis_left/davis_ros_driver/aps_enabled: True
 * /davis_left/davis_ros_driver/dvs_enabled: True
 * /davis_left/davis_ros_driver/exposure: 4000
 * /davis_left/davis_ros_driver/frame_delay: 0
 * /davis_left/davis_ros_driver/imu_acc_scale: 3
 * /davis_left/davis_ros_driver/imu_enabled: True
 * /davis_left/davis_ros_driver/imu_gyro_scale: 3
 * /davis_left/davis_ros_driver/master: True
 * /davis_left/davis_ros_driver/max_events: 0
 * /davis_left/davis_ros_driver/reset_timestamps_delay: 2.0
 * /davis_left/davis_ros_driver/serial_number: 02460074
 * /davis_left/davis_ros_driver/streaming_rate: 30
 * /davis_right/davis_ros_driver/DiffBn_coarse: 4
 * /davis_right/davis_ros_driver/DiffBn_fine: 39
 * /davis_right/davis_ros_driver/OFFBn_coarse: 4
 * /davis_right/davis_ros_driver/OFFBn_fine: 0
 * /davis_right/davis_ros_driver/ONBn_coarse: 6
 * /davis_right/davis_ros_driver/ONBn_fine: 200
 * /davis_right/davis_ros_driver/PrBp_coarse: 2
 * /davis_right/davis_ros_driver/PrBp_fine: 58
 * /davis_right/davis_ros_driver/PrSFBp_coarse: 1
 * /davis_right/davis_ros_driver/PrSFBp_fine: 33
 * /davis_right/davis_ros_driver/RefrBp_coarse: 4
 * /davis_right/davis_ros_driver/RefrBp_fine: 25
 * /davis_right/davis_ros_driver/aps_enabled: True
 * /davis_right/davis_ros_driver/dvs_enabled: True
 * /davis_right/davis_ros_driver/exposure: 4000
 * /davis_right/davis_ros_driver/frame_delay: 0
 * /davis_right/davis_ros_driver/imu_acc_scale: 3
 * /davis_right/davis_ros_driver/imu_enabled: True
 * /davis_right/davis_ros_driver/imu_gyro_scale: 3
 * /davis_right/davis_ros_driver/master: False
 * /davis_right/davis_ros_driver/max_events: 0
 * /davis_right/davis_ros_driver/serial_number: 84010026
 * /davis_right/davis_ros_driver/streaming_rate: 30
 * /dvs_renderer_left/display_method: red-blue
 * /dvs_renderer_right/display_method: red-blue
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    dvs_renderer_left (dvs_renderer/dvs_renderer)
    dvs_renderer_right (dvs_renderer/dvs_renderer)
    image_view_left (rqt_image_view/rqt_image_view)
    image_view_right (rqt_image_view/rqt_image_view)
  /davis_left/
    davis_ros_driver (davis_ros_driver/davis_ros_driver)
  /davis_right/
    davis_ros_driver (davis_ros_driver/davis_ros_driver)

ROS_MASTER_URI=http://localhost:11311

process[davis_left/davis_ros_driver-1]: started with pid [4629]
process[davis_right/davis_ros_driver-2]: started with pid [4630]
process[dvs_renderer_left-3]: started with pid [4631]
process[dvs_renderer_right-4]: started with pid [4637]
process[image_view_left-5]: started with pid [4643]
process[image_view_right-6]: started with pid [4644]
[ WARN] [1644991575.132792035]: Requested serial number: 02460074
2022-02-16 11:36:15 (TZ+0530): CRITICAL: DAVIS ID-1: Failed to open device, no matching device could be found or opened.
[ WARN] [1644991575.181016319]: Could not find DAVIS. Will retry every second.

What might be the problem here and is it possible to record two dvs 346 at the same timestamps?

guillermogb, 2022-02-16:
Hello. Probably you have to change the serial numbers of the devices you want to open, saying which one is the left DAVIS and which one is the right DAVIS.
See the line --> Requested serial number: 02460074

Darshanbhandarics, 2022-02-16:
Thank you, I changed the serial names according to what's written on the cameras, but still the same problem remains the same.

<launch>
  
  <node ns="davis_left" name="davis_ros_driver" pkg="davis_ros_driver" type="davis_ros_driver" output="screen">
    <param name="serial_number" type="str" value="0466" /> 
    <rosparam command="load" file="$(find davis_ros_driver)/config/DAVIS240C.yaml" />
    <param name="master" value="True"/>

    
    <param name="reset_timestamps_delay" value="2.0"/>
  </node>

file: stereo.launch

  
  <node ns="davis_right" name="davis_ros_driver" pkg="davis_ros_driver" type="davis_ros_driver" output="screen">
    <param name="serial_number" type="str" value="0332" /> 
    <rosparam command="load" file="$(find davis_ros_driver)/config/DAVIS240C.yaml" />
    <param name="master" value="False"/>

    
    <remap from="reset_timestamps" to="/davis_left/reset_timestamps"/>
  </node>
</launch>

Terminal:
process[davis_left/davis_ros_driver-1]: started with pid [15090]
process[davis_right/davis_ros_driver-2]: started with pid [15091]
process[dvs_renderer_left-3]: started with pid [15097]
process[dvs_renderer_right-4]: started with pid [15102]
process[image_view_left-5]: started with pid [15104]
process[image_view_right-6]: started with pid [15105]
[ WARN] [1645005477.615563440]: Requested serial number: 0332
[ WARN] [1645005477.631797675]: Requested serial number: 0466
2022-02-16 15:27:57 (TZ+0530): CRITICAL: DAVIS ID-1: Failed to open device, no matching device could be found or opened.
[ WARN] [1645005477.687279019]: Could not find DAVIS. Will retry every second

Also,
When I run the following command: "ROS_NAMESPACE=my_stereo rosrun stereo_image_proc stereo_image_proc" I am getting this warning.
[ WARN] [1645005610.675796275]: The input topic '/my_stereo/left/image_raw' is not yet advertised
[ WARN] [1645005610.676292401]: The input topic '/my_stereo/left/camera_info' is not yet advertised
[ WARN] [1645005610.676307706]: The input topic '/my_stereo/right/image_raw' is not yet advertised
[ WARN] [1645005610.676318308]: The input topic '/my_stereo/right/camera_info' is not yet advertised

guillermogb, 2022-02-16:
The SN is 8-digit. I think you need to pad with zeros or with 0246. Sorry, I cannot check now. Try this, if not, I will check later on the computer that I have this information. -> 02460332 or 00000332

Darshanbhandarics, 2022-02-16:
@guillermogb Thanks a lot! We have to pad with zeros.

One more query, When I run the follwing command:
rosrun camera_calibration cameracalibrator.py --approximate 0.1 --size 8x6 --square 0.025 --no-service-check right:=/my_stereo/right/image_raw left:=/my_stereo/left/image_raw right_camera:=/my_stereo/right left_camera:=/my_stereo/left

I see blank window, but for single camera calib, It is working fine.

when I launch stereo, I see this
ROS_MASTER_URI=http://localhost:11311

process[davis_left/davis_ros_driver-1]: started with pid [27738]
process[davis_right/davis_ros_driver-2]: started with pid [27739]
process[dvs_renderer_left-3]: started with pid [27740]
process[dvs_renderer_right-4]: started with pid [27747]
process[image_view_left-5]: started with pid [27751]
process[image_view_right-6]: started with pid [27753]
[ WARN] [1645018365.947065073]: Requested serial number: 00000466
[ INFO] [1645018365.962557482]: DAVIS ID-1 SN-00000466 [4:5] --- ID: 1, Master: 1, DVS X: 346, DVS Y: 260, Logic: 6.

[ WARN] [1645018365.964225130]: Requested serial number: 00000332
[ INFO] [1645018366.325955989]: using default calibration URL
[ INFO] [1645018366.326077479]: camera calibration URL: file:///home/lenovo/.ros/camera_info/DAVIS-00000466.yaml
[ INFO] [1645018366.326185981]: Unable to open camera calibration file [/home/lenovo/.ros/camera_info/DAVIS-00000466.yaml]
[ WARN] [1645018366.326237172]: Camera calibration file /home/lenovo/.ros/camera_info/DAVIS-00000466.yaml not found.
[ INFO] [1645018366.657593513]: Reset timestamps on DAVIS-00000466 to 1645018366.657561064.
[ INFO] [1645018366.660386810]: Started timer to reset timestamps on master DVS for synchronization (delay=2.00s).
[ INFO] [1645018366.712570903]: DAVIS ID-1 SN-00000332 [3:11] --- ID: 1, Master: 1, DVS X: 346, DVS Y: 260, Logic: 6.

[ INFO] [1645018367.069548882]: using default calibration URL
[ INFO] [1645018367.069646967]: camera calibration URL: file:///home/lenovo/.ros/camera_info/DAVIS-00000332.yaml
[ INFO] [1645018367.069744550]: Unable to open camera calibration file [/home/lenovo/.ros/camera_info/DAVIS-00000332.yaml]
[ WARN] [1645018367.069780484]: Camera calibration file /home/lenovo/.ros/camera_info/DAVIS-00000332.yaml not found.
[ INFO] [1645018367.070104666]: Reset timestamps on DAVIS-00000332 to 1645018367.070080519.
[ INFO] [1645018367.400519057]: Reset timestamps on DAVIS-00000332 to 1645018367.400461197.
[ INFO] [1645018368.660626824]: Reset timestamps on DAVIS-00000466 to 1645018368.660577059.
[ INFO] [1645018368.664312961]: Reset timestamps on DAVIS-00000332 to 1645018368.664267063.

Is it because of discrepancy in timestamp of two cameras after decimal point?