uzh-rpg/rpg_dvs_ros #138: Could not find DVS

gauTT99, 2022-06-14:
Hello, i'm trying to use rpg_dvs_ros on jetson nano with a dvs128. When i run dvs_mono.launch i got this problem. 
: CRITICAL: DVS128 ID-1: Failed to open device, no matching device could be found or opened.
[ WARN] [1655261471.175399543]: Could not find DVS. Will retry every second
I see the dvs when i run lsusb command and the dvs works on others devices (raspberry and linux desktop) so i don't think that is a hadware problem. Does anyone know the solution?

levenberg, 2022-07-11:
Hello @gauTT99 , 

Have you solved this problem? I am using jetson xavier with a dvxplorer camera. When I run the dvs_mono.launch, I have the same problem that cannot find the device.

MoisesFelipe, 2022-09-19:
I'm facing the exact same issue on a Jetson Orin. I'm using the latest dvxplorer camera.

MoisesFelipe, 2022-09-19:
Anyone found a solution for this?